Solid-Energy-Systems/NewareNDA #24: Cycle number calculation vs decoded from data

vividh-garg, 2023-05-09:
Why do you calculate cycle no. when you can just decode it from the data itself as Neware stores the cycle no. natively (as seen in the xls/xlsx files). There are many recipes/tests which involve multiple instances of charge and discharge in the same cycle. So your calculation comes out wrong in those tests.

I am currently using the code below to get the cycle number. Essentially, the correct cycle number here will be 'Cycle +1'. 

```
[Index, Cycle] = struct.unpack('<IH', byte_stream[2:8])
```

d-cogswell, 2023-05-09:
This is interesting, the newest version of Neware BTSDA returns different cycle numbers than older versions for the same nda file. Looks like they switched to reporting the cycle number from the nda. Previously, the only way I could match BTSDA was with `_generate_cycle_number(df)`.

I wonder if the cycle number in the nda is a 4-byte integer?

d-cogswell, 2023-05-12:
I just created a pull request with a flag to switch between the two different cycle numbers. What do you think?

vividh-garg, 2023-05-12:
Let me check if my current code works with old nda files and that too if I have any old nda file. I also found another multiplier value of -12000 which is used in SIM step. Couldn't get to test around with it much.