Solid-Energy-Systems/NewareNDA #27: Multiplier Key Error?

CassiusClark, 2023-07-17:
Hi, having an issue loading cell data for .nda files, receiving the error KeyError: -1.
Just using the df1 = NewareNDA.read(path) command.
Any help appreciated, thanks.
Full error report below.

`---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[49], line 1
----> 1 df1 = NewareNDA.read(path)

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\NewareNDA\NewareNDA.py:28, in read(file, software_cycle_number)
     26 _, ext = os.path.splitext(file)
     27 if ext == '.nda':
---> 28     return read_nda(file, software_cycle_number)
     29 elif ext == '.ndax':
     30     return read_ndax(file)

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\NewareNDA\NewareNDA.py:88, in read_nda(file, software_cycle_number)
     83 if len(bytes) == record_len:
     84 
     85     # Check for a data record
     86     if (bytes[0:2] == b'\x55\x00'
     87             and bytes[82:87] == b'\x00\x00\x00\x00'):
---> 88         output.append(_bytes_to_list(bytes))
     90     # Check for an auxiliary record
     91     elif (bytes[0:1] == b'\x65'
     92           and bytes[82:87] == b'\x00\x00\x00\x00'):

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\NewareNDA\NewareNDA.py:146, in _bytes_to_list(bytes)
    143 if Index == 0 or Status == 0:
    144     return []
--> 146 multiplier = multiplier_dict[Range]
    148 # Create a dictionary for the record
    149 list = [
    150     Index,
    151     Cycle + 1,
   (...)
    161     datetime(Y, M, D, h, m, s)
    162 ]

KeyError: -1`

d-cogswell, 2023-07-19:
@CassiusClark alternatively, I can explain how to find the range multiplier if you can reproduce using the Python debugger.

vividh-garg, 2023-07-22:
It must be the SIM step or the Pulse step in the test.

d-cogswell, 2023-08-07:
@CassiusClark this is probably an easy fix if you can share a little more information.

CassiusClark, 2023-09-08:
Hi, sorry for the delay, I can email you over a file. Thanks.

d-cogswell, 2023-09-08:
0d0a701 should fix this.

closed 2023-10-09 by d-cogswell