Solid-Energy-Systems/NewareNDA #31: Reading aux data with NewareNDA as well

Grimler91, 2023-09-15:
This issue hopes to shed some light on the format of auxiliary data in the ndax files, and to track progress in making NewareNDA read the data.

In the equipment I am working with is possible to have extra/auxiliary voltage and temperature measurements. In BTS settings it is then possible to select whether this data should be exported, and whether it should be merged with the main data or put in a separate excel sheet (when exporting to excel):
[image]

In the ndax archive the data then contains one data_Aux.ndc file, and one file per aux channel named something like YYYYMMDD_hhmmss_#_#_#_#_1.ndc, where the last digit is increased for each file.
Probably data_Aux.ndc contains info about how many aux channel files there are, and/or about their format, and the other ndc files contain the actual data.

Has anyone had any success in understanding the format of these aux ndc files?

d-cogswell, 2023-09-18:
Hi @Grimler91 I only recently added ndax support and haven't added Aux channel support yet. The temperature data is somewhere in the ndc files.

Grimler91, 2023-09-18:
Hi @d-cogswell, that's fine. 
I'll update this issue with documentation if I get some time to investigate the format of those extra ndc files.

d-cogswell, 2023-09-18:
It looks to me like the temperature data is stored in the ndc files with timestamp names.

Grimler91, 2023-09-18:
Yeah, I think so too, removing those files removes the corresponding aux channel from BTSDA when opening the main ndc file. 

However, the  data_aux.ndc file also contains some sort of necessary data, removing that file removes all aux data when opening main ndc with BTSDA.

d-cogswell, 2023-09-20:
Hi @Grimler91 I may have a solution here [issue/29](https://github.com/Solid-Energy-Systems/NewareNDA/tree/issue/29). It seems to be working correctly for me but I don't have very many ndax files with temperature for testing. What do you think?

Grimler91, 2023-09-25:
@d-cogswell Tested your patch with an ndax file with 3 temperature sensors connected. NewareNDA correctly identified all three sensors and gave accurate data, so looks good to me!

I also have aux voltage data in my logs, will have a look at if I can parse that data as well. 

If you are interested and think it would be useful I could share an example log in ndax and xlsx in private somewhere.

d-cogswell, 2023-09-25:
Great, thanks for the help. I don't have any files with aux voltage. Can you tell if it's stored in another ndc file? If you are willing to email files to cogswell@mit.edu I can take a look. I'll only use them for testing.

d-cogswell, 2023-10-26:
Implemented in 24957c0

closed 2023-10-26 by d-cogswell