Solid-Energy-Systems/NewareNDA #36: EOFError, no valid records and empty 'Step' Series

AdamOpps, 2023-10-11:
I am trying to use the CLI script to export .nda files from a Neware BTS9000 to excel and have run into a few errors. On first attempt, I was getting the EOF-Error, stating that there are no valid records. Upon binary file inspection, the file did not have the specified identifier contained in it. 

Based on my best guess, I set the header length to be a static 1024 bytes and bypassed the header loop. Doing this raised an IndexError in _count_changes, thus the data series (df['Step']) passed was empty. It looks like my .nda file does not save data in the same structure than this script expects. The .nda file isn't empty, as I can load it back into the BTS client (version 9.0.x.x) on a different computer and the data is present.

Any ideas of what might be going on? Is there some software setting in BTS/BTS9 that I should be using? I can send an example .nda file or two if you are willing to look into it.

Thanks!

d-cogswell, 2023-10-12:
Interesting, I don't have access to any BTS9000 so I'm not sure what might be different. Are you able to share some examples? If so you can email to cogswell@mit.edu.

d-cogswell, 2023-11-14:
@AdamOpps I think I found a solution. See #40. Can you help with some testing?

AdamOpps, 2023-11-14:
Yes, I have some time this week that I can help out! Anything specific that you need from me, or just to run a bunch of files through the new version?

AdamOpps, 2023-11-14:
I've tried a few different files and they seem to work!

d-cogswell, 2023-11-15:
Great, thanks. Check out the latest commit, which separates the discharge and charge capacity and energy. I'm running some regression testing on my side, and then I think the change are ready to merge.

AdamOpps, 2023-11-15:
I'm struggling to get the new commit changes. The terminal tells me that I am both up to date, and unable to resolve a Pull Request with the number 40. Any ideas? My Git knowledge ends past pushing and pulling my own changes to access on different computers.

On another note, I tried extracting data from another .nda file from August and am coming up with a key error of -10 in the multiplier_dict. Should I create another issue, or just email you the file again?

d-cogswell, 2023-11-16:
@AdamOpps you should be able to `git pull origin issue/36`. Does that work?

You are welcome to commit your additional multiplier values. Thanks for the help. If you email me the nda file, I can double-check that it's good to go.

AdamOpps, 2023-12-04:
I am trying to push my commits to dicts.py so I can close this issue, but I am denied permission (error 403) to 'origin issue/36'. Is there a request protocol that I am missing?

Further, I have some other changes out of scope of this issue that I'd like to submit for review. I imagine I'll have a similar block there?

d-cogswell, 2023-12-05:
Hi @AdamOpps I just sent you an invite to this repo so you can push your commits. This should fix the permission problem.

In general if you don't have permissions, the thing to do it create a fork, commit your changes there, and then create a pull request:
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

AdamOpps, 2023-12-14:
I can't create a new branch without some permission, I've been trying to go about this and am making no progress. Could you resend the invitation as it has expired. Thanks!

d-cogswell, 2023-12-15:
@AdamOpps I just resent the invitation. I also need to do some reorganization. The code for the new and old nda versions should be moved to separate functions. I tried to find one solution that worked for everything, but some of my regression tests on old files are failing.

d-cogswell, 2023-12-18:
I made my changes to keep the old code in place for old nda. How about if I merge the branch? You can aways create a new branch for other changes.

AdamOpps, 2023-12-18:
Okay, I finally was able to add the extra keys in dicts.py and push them. I think this issue is resolved, so go ahead and merge and close the issue. Thanks for all of your help and patience with this!

closed 2024-01-03 by AdamOpps