Solid-Energy-Systems/NewareNDA #37: Is it possible to export "cycle" data only

nevergow, 2023-10-21:
When transfer ndax data into excel, there will be several sheets, like "record", "cycle" and so on. So, is it possible to extract "cycle" data, in which only contain data of capacity, energy, start time, end time and any other data ,just like in excel from ndax file by this script?

d-cogswell, 2023-10-21:
Hi @nevergow, the data on those extra Excel tabs is not stored in the ndax file itself. Instead it's calculated from the raw data during the export process. The goal of this code is just to extract the data. If you need cycle summary statistics, it's not too hard to generate them yourself using [pandas.DataFrame.groupby](https://pandas.pydata.org/docs/reference/api/pandas.DataFrame.groupby.html). Hope that helps!

nevergow, 2023-10-21:
oh, I see, thank you!

closed 2023-10-30 by d-cogswell