Solid-Energy-Systems/NewareNDA #42: KeyError due to missing element in multiplier_dict

SGallagherGelion, 2023-12-06:
Hello,
I am developing a script to read neware nda files into my database.
I have encountered the following error due a missing key/value in the multiplier dict.

Stacktrace:
```
2023-12-06T15:27:08.923267 [INFO] NewareNDA.read_nda: NDA version: 29
2023-12-06T15:27:08.923267 [INFO] NewareNDA.read_nda: Server: BTSServer(R3)-7.6.0.301 (2018.07.30)
2023-12-06T15:27:08.923267 [INFO] NewareNDA.read_nda: Client: BTS Client 8.0.0.474(2020.12.22)
Traceback (most recent call last):
  File "C:\Users\User\projects\project\src\prototype\main.py", line 51, in <module>
    main()
  File "C:\Users\User\projects\project\src\prototype\main.py", line 43, in main
    df = NewareNDA.read(temp_file)
         ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\projects\project\venv\Lib\site-packages\NewareNDA\NewareNDA.py", line 30, in read
    return read_nda(file, software_cycle_number)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\projects\project\venv\Lib\site-packages\NewareNDA\NewareNDA.py", line 94, in read_nda
    output.append(_bytes_to_list(bytes))
                  ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\projects\project\venv\Lib\site-packages\NewareNDA\NewareNDA.py", line 152, in _bytes_to_list
    multiplier = multiplier_dict[Range]
                 ~~~~~~~~~~~~~~~^^^^^^^
KeyError: 1
```

Possible Solutions:
1. Add element to for key 1 with value 1e-4
This appears to be the correct multiplier for my example based off my own testing.
Unfortunately, I am not able to share the data files as it is confidential.

2. Set multiplier via keyword argument  
df = NewareNDA.read('my_file.nda', multiplier={1:1e-4})

Please advise on next steps to move this forward.

Regards, 
Steven Gallagher

snoehmer, 2023-12-06:
Hi,
I also found a missing entry in the `multiplier_dict` that is used by the CT-4008Q series:
`-100000000: 10`

That is the only missing range I found when importing multiple ndax files from BTS server 8.0. That entry could be added as well if you are already onto it.

Thanks for the great work on this project, helps a lot!

BR
Stefan

d-cogswell, 2023-12-06:
I just added these ranges to the development branch. Thanks for your help! You can install the development branch with this pip command:

`pip install git+https://github.com/Solid-Energy-Systems/NewareNDA.git@development`

snoehmer, 2023-12-06:
@d-cogswell Great, thanks!

SGallagherGelion, 2023-12-08:
@d-cogswell thankyou for a fast response!
I have tested the development branch and it looks like it solves my problem.

closed 2023-12-19 by d-cogswell