Solid-Energy-Systems/NewareNDA #44: Issue with interpolating data from ndax file

lithiosEngineering, 2023-12-19:
Hi, having an issue with interpolating missing data from ndax files.  Have not modified any of the original code.  Just installed the package and am trying to run the example usage (import NewareNDA --> df = NewareNDA.read('testndax.ndax').  Consistently seeing the following issue: 

----------------------------------------------------------------------------------------------------------------------
/usr/local/lib/python3.10/dist-packages/pandas/core/frame.py in interpolate(self, method, axis, limit, inplace, limit_direction, limit_area, downcast, **kwargs)
  11853         **kwargs,
  11854     ) -> DataFrame | None:
> 11855         return super().interpolate(
  11856             method,
  11857             axis,

TypeError: super(type, obj): obj must be an instance or subtype of type

-------------------------------------------------------------------------------------------------------------------------

triggered by this line in the _data_interpolation(df) function

-------------------------------------------------------------------------------------------------------------------------

[/usr/local/lib/python3.10/dist-packages/NewareNDA/NewareNDAx.py](https://localhost:8080/#) in _data_interpolation(df)
    103 
    104     # Group by step and run 'inside' interpolation on Time
--> 105     df['Time'] = df.groupby('Step')['Time'].transform(
    106         lambda x: pd.DataFrame.interpolate(x, limit_area='inside'))
    107

-----------------------------------------------------------------------------------------------------------------------------

any help is appreciated.  thanks!

d-cogswell, 2023-12-19:
Hi @lithiosEngineering what version of pandas are you using? Can you try upgrading to the latest version? I'm noticing this error goes away starting in pandas 2.1.0. I will investigate a workaround for older versions of pandas. The code in NewareNDA causing this error is new and has not had much testing.

d-cogswell, 2023-12-19:
This might be a quick fix for older pandas versions. Can you test the latest commit to development and see if the issue is fixed? 6f28da8

lithiosEngineering, 2023-12-19:
Thanks Dan that works!

closed 2024-01-12 by d-cogswell