Solid-Energy-Systems/NewareNDA #48: Missing test data and CI

ml-evs, 2024-02-25:
Related to #47 and an upcoming PR, there are limited contribution instructions, and it is currently impossible to run the integration tests as the test data is not provided in this repo.

Also, there is no continuous integration testing of new PRs (via e.g., GitHub actions) which can go a long way to prevent contributions and also make expectations clear.

I'm happy to help set this up (it is 5 mins work once you've done it before) if you like, but probably having the test data available is a minimum requirement!

d-cogswell, 2024-02-26:
Hi @ml-evs these are good suggestions. The main issue with finishing off CI is I haven't been able to find public data to use. I run a lot of regression tests, but it's on private company data which I'm not able to share publicly.

Any ideas?

ml-evs, 2024-02-26:
That's fair! Well, we have some ndax files that can probably be made public (I can ask the owners). There's also .nda files from the other open source neware reader package, whilst it's open it might be nice to reach out to them and ask (and maybe link to each other's packages in the respective READMEs). I'm not imagining an extensive test set (though it's great you've got more testing happening behind the scenes) but just a single example of each file type would be a good start. 

I'm also involved in a more meta initiative https://github.com/marda-alliance/metadata_extractors with a registry of file types and extractors that could join some of these dots, happy to add this repo to the list over there unless you object!

Grimler91, 2024-02-28:
Hi, I can check with the company I'm working for if they would be open to sharing some data as well. 

As @ml-evs said I guess files for each type would be a good start. Thinking larger, do we want to have data for as many step types, current/voltages ranges and instrument models as possible? (Any other parameters/settings that could be important?)

I guess it would preferred to have many small datafiles, rather than having a large one with many different types of steps?

d-cogswell, 2024-02-29:
I plan to start working on CI next, starting with the file from neware_reader. CI will eventually need a collection of nda/ndax files of different versions, and ideally some with auxiliary data, and some without. Code coverage will identify what additional files are needed.

One idea is to generate public test data from different hardware using commercial batteries (NiMH, alkaline?) so that no sensitive company data is revealed. I will ask for permission to do this.

d-cogswell, 2024-03-06:
I got a Github workflow running and added a publicly available nda file:
https://github.com/Solid-Energy-Systems/NewareNDA/tree/workflow

closed 2025-02-13 by d-cogswell