Solid-Energy-Systems/NewareNDA #50: Some aux data missing from (some) files

Grimler91, 2024-03-01:
Hi, 

I have noticed that NewareNDA sometimes fails to extract all aux data, and instead just returns NaN after some point. Non-aux values work fine. Here is an example plot with capacity and aux temperature plotted with data from either an exported xlsx file, or from NewareNDA:

[image]

As can be seen the aux temperature is missing in NewareNDA data after roughly cycle index 250.
(I will send the data to Dan in private)

d-cogswell, 2024-03-04:
The aux data seems to be stored in ndc version 5, which I don't think I've seen before. Does this come from a different machine, or did you upgrade software or something?

Grimler91, 2024-03-05:
Same machine, but I upgraded bts server & client in middle of November, to "BTS 8.0.1 20230531". Data I sent previously must have been with previous version, so that likely explains issues. 

It is rather impressive then instead that newareNDA can (almost) handle a ndc version it was not made for :)

d-cogswell, 2024-03-08:
I pushed a commit to development that I hope fixes this issue. @Grimler91 can you do some testing and let me know if you find any issues?

Grimler91, 2024-03-14:
Haven't discovered any issues so far, development branch works well with ndax files I have, thanks!

d-cogswell, 2024-03-29:
Fixed in 973bb54

PR #54 (2024-04-01): v2024.04.01
Monthly update contains:
Completed Github workflow with regression testing and code coverage (more test cases needed!)
New hardware ranges
Bug fix #50
Code cleanup
closed 2024-03-29 by d-cogswell