Solid-Energy-Systems/NewareNDA #56: Query Regarding Data Extraction from NDAX Files and Specific Capacity Information

Naturalmys, 2024-03-28:
May I kindly inquire whether it's feasible to extract information regarding the cycling layers from the ndax file? Additionally, I'm curious if the extracted data pertains to overall capacity rather than specific capacity. Would it be possible to obtain data in terms of specific capacity? thanks !

d-cogswell, 2024-03-28:
Hi @Naturalmys thanks for the question. The cycling layers are added by Neware in software post-processing. The layers don't exist in nda/ndax, but you can generate them using Pandas [groupuby()](https://pandas.pydata.org/pandas-docs/stable/reference/api/pandas.DataFrame.groupby.html). The goal of NewareNDA is to read from files without getting into post-processing.

The capacities being returned are nominal capacity, which is what's stored in the files. To get specific capacity you should divide these values by the loading.

AdamOpps, 2024-04-01:
Along these lines, is the test metadata saved in these .nda/x files? Things like the mass and remarks would be valuable to extract if it did exist in these files.

Grimler91, 2024-04-03:
> Along these lines, is the test metadata saved in these .nda/x files? Things like the mass and remarks would be valuable to extract if it did exist in these files.

Not sure about nda, and not sure about mass, but remarks and other metadata can be found in ndax files in different human readable .xml files (called TestInfo.xml, Step.xml, VersionInfo.xml).

For reference you can unzip an ndax and check the content (run something like `unzip /path/to/file.ndax` from command line, or rename from the ndax file to zip and then extract with some zip program)

AdamOpps, 2024-04-10:
That is interesting and good to know! Unfortunately, I am unable to get access to .ndax files from our cycling experiments. I'm unsure whether that's due to hardware or software limitations, but we tend to deal in just .nda files.

Regardless, I have begun making progress in locating these items in the .nda hex file. I have found the remarks, and it looks like the mass (or at least some value related to the mass) may be saved with each data point. I'll make a dedicated post and perhaps some code proposition for extracting these values in the near future.

It appears that specific capacity should be attainable in some way, but it may be dependent on hardware or software versions.

d-cogswell, 2024-04-12:
Hey @AdamOpps that's interesting. There are still bytes in each record which I haven't been able to identify, which could be mass. We typically don't define a loading, so if mass is being stored somewhere I would have a tough time finding it.

AdamOpps, 2024-04-17:
**I have fount the Active Mass information!**
@d-cogswell Check out #58 where I detail its location, how to extract it, and implementation suggestions. Also, let me know if there is a better place for me to post notes/discussion for non-bug-related topics in the future!

I also have evidence that suggests the unidentified bytes might be Differential Capacity (dQ/dV potentially). That is to be determined.

closed 2024-06-03 by d-cogswell