Solid-Energy-Systems/NewareNDA #60: BTS85 charge/discharge capacity values are unrealistic

vajeeston, 2024-05-15:
Hi there!

Could you assist me in overcoming this issue? When I run the NewareNDA on the latest files generated from BTS85, the charge/discharge capacity values are unrealistic. However, when we open these files in BTSDA 8.0.0.526, the values appear normal (please refer to the attached figure). For your convenience, I have attached an example file to this message. I am eager to resolve this issue.
[image]
[BTS85-36-6-5-110-20240424.zip](https://github.com/Solid-Energy-Systems/NewareNDA/files/15322177/BTS85-36-6-5-110-20240424.zip)

d-cogswell, 2024-05-15:
Hi @vajeeston it looks like you have a new file version that I haven't seen before. Thanks for sharing the file. It's probably easy to fix, but can you direct me to a version of BTSDA that will open this file?

vajeeston, 2024-05-16:
Hi! 
As I have mentioned, this can open in BTSDA 8.0.0.526 and other versions. I have checked, and it works well with previous versions as well. Thank you for your support!

d-cogswell, 2024-05-16:
Thanks, I got the file to open after installing BTS 8.0.1. It's a new version of ndc with the data packed slightly differently. I'm working on it.

d-cogswell, 2024-05-16:
I think I have a solution in this pull request: #62

vajeeston, 2024-05-16:
Is it possible to get the code?

d-cogswell, 2024-05-16:
Yes, you can download the branch with this link:
https://github.com/Solid-Energy-Systems/NewareNDA/archive/refs/heads/issue/60.zip

vajeeston, 2024-05-21:
Hi! 

Now the voltage values are correct, but, the charge and discharge capacity values are not correct. please see the figure. 
[image]

d-cogswell, 2024-05-22:
Hi @vajeeston  I checked the results from NewareNDA again and they look correct to me. You should compare either individual records, or a grouping by step number such as:
`df.groupby(['Cycle', 'Step']).last()`

Comparing by cycle number can be confusing because Neware has several different options for how it assigns cycle numbers. I think you can reproduce what you are seeing in BTSDA by loading your file like this:
`df = NewareNDA.read('BTS85-36-6-5-110-20240424.ndax', software_cycle_number=False)`

vajeeston, 2024-05-23:
Thank you for your support! it's working fine!

closed 2024-05-23 by d-cogswell

vajeeston, 2024-06-02:
> Hi @vajeeston<https://github.com/vajeeston> would it be ok with you if I use your ndax file from this issue as a regression test for NewareNDA?

Hi Dan,

Ok no problem!