Solid-Energy-Systems/NewareNDA #61: Read a value that does not appear in the original NDAX file

ImpossibleMissi0N, 2024-05-16:
Hello,

When I use this useful tool to read the NDAX file, I find one and only one row value is not aligned with the original NDAX file. 
[image]

As is shown in this picture underlined by green line, the Time should be 0.0 instead of 12.0, Step should be 36 instead of 35. You can also refer to the original file screenshot. 
[image]

Besides, when I read the NDAX file I got the warning " UserWarning: IMPORTANT: This ndax has missing data. The output from NewareNDA contains interpolated data! warnings.warn("IMPORTANT: This ndax has missing data. The output from " " The version of the BTS version is 8.0.0.526

Best regards,
Aaron

vajeeston, 2024-05-16:
Hi! 
I hope the without data_interpolation_function will solve this problem.  

Try this:

Just comment out this line in the NewareNDAx.py file:
_data_interpolation(data_df)

ImpossibleMissi0N, 2024-05-16:
> Hi! I hope the without data_interpolation_function will solve this problem.
> 
> Try this:
> 
> Just comment out this line in the NewareNDAx.py file: _data_interpolation(data_df)

yes it indeed sloved the warning, but too many points have been the NaN and I cannot use the data and the data is still not same as the NDAX file when I use BTSDA to check. So, I think I need another solution. Thanks！
[image]

d-cogswell, 2024-05-17:
Hi @ImpossibleMissi0N after turning off data interpolation, do the rows with data match what you see in BTSDA?

The warning is there for a good reason. In recent file versions, Neware does not seem to store all of the data. I suspect some data in BTSDA is being generated by interpolation rather than being stored in the file.

ImpossibleMissi0N, 2024-06-04:
> Hi @ImpossibleMissi0N after turning off data interpolation, do the rows with data match what you see in BTSDA?
>
> The warning is there for a good reason. In recent file versions, Neware does not seem [...]

Thanks for your reply. Actually no, I think you are right. Some data is generated by interpolation. But when I read use the python, I got the some rows that are not same as displayed in the BTSDA. (As I put in the first thread picture).  And it often show one more row, for example, I set pulse as 25 but before this row, I can find a row with pulse is 0. This cannot be found in the BTSDA, but it will come up in the .ndax file.

d-cogswell, 2024-06-05:
Hi, I wonder if the issue is how the cycle number is being assigned. Neware has a few different ways of doing this. Do your results match if you load your file like this?

`NewareNDA.read('file.ndax', software_cycle_number=False)`

ImpossibleMissi0N, 2024-06-06:
Not really. But now I have used a more verbose way to manipulate the data using count the times of each cycle and only select the specified numbered rows. Now it can work even if it is not that convenient. Thanks

d-cogswell, 2024-06-06:
Hi, can you email me your file? I'd like to investigate this if possible.

d-cogswell, 2024-07-01:
Hi @ImpossibleMissi0N were you able to figure out where the discrepancy is? Can you share a file?

closed 2024-08-14 by d-cogswell