Solid-Energy-Systems/NewareNDA #72: NewareNDA Extraction Issue: Only One Row Extracted with Incorrect Values

ferdaoues5, 2024-06-06:
Hello, I'm having an issue with the execution of NewareNDA. It only extracts a single line, and its values are incorrect. I'm currently working on a .nda file version 9 (130).
[image]

d-cogswell, 2024-06-06:
Hi @ferdaoues5 , it's hard to say what might be causing this. Are you able to share you file so I can investigate? If so, you can email it to cogswell@mit.edu.

d-cogswell, 2024-06-10:
Hmm, it looks like BTS9.1 introduces a new file format that is different than BTS9.0. It has the same file version which is why NewareNDA is trying and failing to read it. I'll see what I can do, I may need some help testing because I don't have many BTS9000 files.

d-cogswell, 2024-07-01:
I'm going to close this, but please reopen if you notice any problems.

closed 2024-07-01 by d-cogswell