Solid-Energy-Systems/NewareNDA #85: Accessing data_log.ndc and data_es.ndc

ianrnorthvolt, 2024-08-09:
Hi! This is not so much an "issue", but a request for help.

I am giving Neware the benefit of the doubt and really hoping that they are not interpolating data. I wonder if the missing data is contained in the data_log.ndc and data_es.ndc files? These are currently not converted from binary in this package.

I tried to convert these by adapting the functions that you have written for the other files, but I am not sure what the numbers you have used in the conversions should be e.g. Record length, header, and the numbers used [here](https://github.com/Solid-Energy-Systems/NewareNDA/blob/master/NewareNDA/NewareNDAx.py#L341) and [here](https://github.com/Solid-Energy-Systems/NewareNDA/blob/master/NewareNDA/NewareNDAx.py#L384).

How did you find these numbers when you were creating the existing functions?

Have you tried to convert these other ndc files in the later versions of the software that have the missing data issue talked about [here](https://github.com/Solid-Energy-Systems/NewareNDA/issues/79)?

d-cogswell, 2024-08-12:
Hi, the data_log.ndc and data_es.ndc files seem to be small, typically <100kb. I'm not sure what's in them, but it doesn't seem like enough space to store the missing data (which for some of my files should be >10MB).

The numbers you referred to come from using a hex editor to look for repeated patterns in the binary data. They are memory offsets, field lengths, starting positions, etc. If you look at data_es.ndc, it's mostly zeros...

The files in question are generally much smaller than similar experiments from other machines, which is a clue. The second clue is if you plot the output of BTSDA, you will see the expected level of noise from the instrument in the voltage and current (all these points are saved in the file). However, the capacity, which I claim is not being saved, will be perfectly linear with no noise. If I make the same plot for an older file, the file will be larger and there is noise in the capacity field.

I suspect this is related to the middle machine, and I first noticed it with version 4S_4.2.5.0_20230308_185745.

ianrnorthvolt, 2024-08-13:
I think the capacity not being saved is ok, we could deal with that! It is the timestamps not being saved that are truly frustrating in some circumstances (mainly the beginning and end of steps).
Thanks for the reply anyway, I appreciate it!

d-cogswell, 2024-08-13:
They seem to save a few time values at the beginning and end of most steps. Maybe there is a setting somewhere that controls this?

Grimler91, 2024-09-10:
Hi, for the record data_log.ndc seems to contain information about alarms or warnings during the test. BTSDA shows a little red light symbol in data where alarms happened: 

[image]

If I unpack an .ndax file, remove data_log.ndc, re-pack it and open it, then those symbols no longer shows in BTSDA.

I do not think I have seen any data_es.ndc files, I guess those are specific to some equipment or BTS versions.

closed 2025-03-11 by ianrnorthvolt