Solid-Energy-Systems/NewareNDA #94: Unable to read files from a new Neware unit

ConnorM-Gaussion, 2024-11-19:
Hi guys,

I'm having some issues reading data files from a new Neware unit. When reading files, various "WARNING:root:Unknown record type: XX" errors are returned, the voltage is scaled incorrectly, the capacity and energy values are wrong and the auxiliary channel data is missing. 

I haven't encountered any of these issues with data files from our other (older) units, so I suspect that this new unit is formatting the ndax file differently. Using zipfile.PyZipFile (which another user used in a different issue thread) it does appear there are at least naming convention differences:
Old Unit File:
['EditStep.xml', 'Step.xml', 'TestInfo.xml', 'VersionInfo.xml', 'data.ndc', '**data_1.ndc**', '**data_2.ndc**', 'data_step.ndc', 'data_runInfo.ndc', 'data_log.ndc', 'data_es.ndc']
New Unit File:
['EditStep.xml', 'Step.xml', 'TestInfo.xml', 'VersionInfo.xml', 'data.ndc', '**data_AUX_10_3_1.ndc**', '**data_AUX_9_3_1.ndc**', 'data_step.ndc', 'data_runInfo.ndc', 'data_log.ndc', 'data_es.ndc']

I've attached a example data file (renamed as a .txt to allow uploading to github) if you'd like to have a look:
[123456789012_Unit27_Example_Data_File_txt.txt](https://github.com/user-attachments/files/17800613/123456789012_Unit27_Example_Data_File_txt.txt)

Any support to resolve this issue would be greatly appreciated! 

Thanks all!

d-cogswell, 2024-11-19:
Thanks for reporting this and for sharing a file. The error is `NotImplementedError: ndc version 14 filetype 5 is not yet supported!`. This is the first file I've seen of the latest ndc version to include aux data, so I need to write a new function to read the binary aux data. I'll let you know when I think I have something working.

ConnorM-Gaussion, 2024-11-20:
Thanks Dan!

d-cogswell, 2024-11-25:
Hi @ConnorM-Gaussion I think I have a solution for reading the temperature data from your file (#95). Can you help do some more testing?

Here is how you can install the update:
`pip install git+https://github.com/Solid-Energy-Systems/NewareNDA.git@ndc_14_filetype_5`

ConnorM-Gaussion, 2024-11-28:
Hi @d-cogswell, I've tested the update on various files from the new Neware unit and it seems to be working very well! The only difference in behaviour I've noticed is occasionally the following "performance warning" message will appear in the console after reading particular files: 

C:\Users\ConnorMcArdle\anaconda3\envs\Neware\Lib\site-packages\NewareNDA\NewareNDAx.py:158: PerformanceWarning: Adding/subtracting object-dtype array to TimedeltaArray not vectorized.
  timestamp = df['Timestamp'].ffill() + \

But it doesn't seem to affect the data read.
Thanks for addressing these compatibility issues so quickly!

d-cogswell, 2024-11-29:
Thanks for the update, I'll go ahead and merge this and create a new release. Is it ok with you if I add the ndax file you shared to the regression test suite?

I'm aware of that `PerformanceWarning` (#86) but I don't have a solution yet. Pandas sometimes doesn't identify the dtype of the date field correctly.

closed 2024-12-06 by d-cogswell