Solid-Energy-Systems/NewareNDA #96: Different timestamp formats for different NDC versions

Grimler91, 2024-12-05:
When reading most of the files I have access to, which typically have NDC version: 5 filetype: 1 or 5, the Timestamp column has a format like: 
```
2024-07-29 17:04:44
```

When reading a file from newer equipment with NDC version: 11 filetype: 1 or 7 or 18, I instead get:
```
2024-07-29 11:38:38+02:00
```

I can't really see why the format is different in the latter case (Timestamp handling in the different read_ndc functions look the same to me). Is it possible to use the same Timestamp format for all ndc versions?

Grimler91, 2024-12-06:
We moved from summer time to normal time during the test, and the time does go from `+02:00` to `+01:00`, so maybe that has to do with it.. But still no idea why it only seems to happen for the test results for this particular cycler.

d-cogswell, 2024-12-09:
Hi @Grimler91 this sounds like a time zone issue. In ndc version 5, the date is stored in a time-zone-unaware format:
https://github.com/Solid-Energy-Systems/NewareNDA/blob/485671c10cb1d1f2498f9623703818d5051b2e6b/NewareNDA/NewareNDAx.py#L515

But in ndc version 11, the dates are stored as Unix timestamps which must be converted to a time zone:
https://github.com/Solid-Energy-Systems/NewareNDA/blob/485671c10cb1d1f2498f9623703818d5051b2e6b/NewareNDA/NewareNDAx.py#L401

The timestamp is read in UTC, and then converted to the time zone of the local computer. Things can get confusing if the computer is in a different time zone from the cycler, or around daylight savings.

It's possible to add or remove the time zone data from the dates to make them consistent:
https://docs.python.org/3/library/datetime.html#datetime.datetime.astimezone

Is the output of NewareNDA consistent with BTSDA?

Grimler91, 2024-12-20:
Aha! Thanks for the insights!

BTSDA is somehow aware of timezone of computer where test was run. NewareNDA gives:
```
0   2024-07-29 17:09:12
1   2024-07-29 17:09:42
2   2024-07-29 17:10:12
Name: Timestamp, dtype: datetime64[ns]
```
and
```
0   2024-07-29 10:37:38+01:00
1   2024-07-29 10:38:38+01:00
2   2024-07-29 10:39:38+01:00
Name: Timestamp, dtype: datetime64[ns, CET]
```
While if I open the same files in BTSDA I see: 

[image]
[image]

Timezone settings seems like something I can try to track down in the .ndax file..

Grimler91, 2025-07-09:
Closing, let's track in github.com/d-cogswell/NewareNDA

closed 2025-07-09 by Grimler91