Solid-Energy-Systems/NewareNDA #97: multiplier_dict KeyError: 500

pablo-voltaiq, 2024-12-16:
Hi, I am trying to load some ndax data and am getting the error below. Would the proper key/value be `500: 1e-1`? 
```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[2], line 1
----> 1 df = nda.read('Neware_8-0-X_Symmetrical_Cell.ndax')

File /opt/conda/lib/python3.9/site-packages/NewareNDA/NewareNDA.py:51, in read(file, software_cycle_number, cycle_mode, log_level)
     49     return read_nda(file, software_cycle_number, cycle_mode)
     50 elif ext == '.ndax':
---> 51     return read_ndax(file, software_cycle_number, cycle_mode)
     52 else:
     53     logger.error("File type not supported!")

File /opt/conda/lib/python3.9/site-packages/NewareNDA/NewareNDAx.py:78, in read_ndax(file, software_cycle_number, cycle_mode)
     76 if 'data.ndc' in zf.namelist():
     77     data_file = zf.extract('data.ndc', path=tmpdir)
---> 78     data_df = read_ndc(data_file)
     79 else:
     80     raise NotImplementedError("File type not yet supported!")

File /opt/conda/lib/python3.9/site-packages/NewareNDA/NewareNDAx.py:203, in read_ndc(file)
    201 try:
    202     f = getattr(sys.modules[__name__], f"_read_ndc_{ndc_version}_filetype_{ndc_filetype}")
--> 203     return f(mm)
    204 except AttributeError:
    205     raise NotImplementedError(f"ndc version {ndc_version} filetype {ndc_filetype} is not yet supported!")

File /opt/conda/lib/python3.9/site-packages/NewareNDA/NewareNDAx.py:272, in _read_ndc_5_filetype_1(mm)
    270     for i in struct.iter_unpack('<87s', bytes[125:-56]):
    271         if i[0][7:8] == b'\x55':
--> 272             output.append(_bytes_to_list_ndc(i[0]))
    274 # Postprocessing
    275 df = pd.DataFrame(output, columns=rec_columns)

File /opt/conda/lib/python3.9/site-packages/NewareNDA/NewareNDAx.py:500, in _bytes_to_list_ndc(bytes)
    497 [Y, M, D, h, m, s] = struct.unpack('<HBBBBB', bytes[75:82])
    498 [Range] = struct.unpack('<i', bytes[82:86])
--> 500 multiplier = multiplier_dict[Range]
    502 # Create a record
    503 list = [
    504     Index,
    505     Cycle + 1,
   (...)
    515     datetime(Y, M, D, h, m, s)
    516 ]

KeyError: 500
```

d-cogswell, 2024-12-17:
Hi @pablo-voltaiq it's best not to guess at these missing range values. The way to get the correct value is with the debugger. When the missing value throws an error, get the values of `Index` and `Current`, and compare against the output of BTSDA at the same `Index`.

If `Current` is zero this is a bit trickier. I set the missing range to 1, and stop the debugger at the first missing range entry with a non-zero current.

Are you able to share your file?

d-cogswell, 2024-12-17:
I will try to write some code which tells you which Index to find in BTSDA when this error is encountered.

d-cogswell, 2024-12-17:
Try installing this debugging branch I just made. It should tell you in the error message which index to find in BTSDA:
https://github.com/Solid-Energy-Systems/NewareNDA/tree/debug/multiplier

pablo-voltaiq, 2024-12-17:
Thanks for the help! I found the current at the datapoint where it errored out: `-0.15001` with units A, I exported the ndax file to xlsx and found it at the datapoint 27876. From the debugger the value I got for the current at that index was: `-15001`.

d-cogswell, 2024-12-17:
Thanks for checking. Since NewareNDA reports units in mA, it looks like the multiplier should be 1e-2.

d-cogswell, 2024-12-17:
I added the multiplier to the development branch [64a2cf4](https://github.com/Solid-Energy-Systems/NewareNDA/commit/64a2cf40b5421ca1b2de7909f3e01245129982aa)

pablo-voltaiq, 2024-12-18:
Thank you so much!

closed 2025-01-02 by d-cogswell