Solid-Energy-Systems/NewareNDA #100: KeyError: 22

AndreSihm, 2025-01-16:
I'm getting the following error when reading certain .ndax files:

>   File C:\ProgramData\conda-venvs\py39\lib\site-packages\NewareNDA\NewareNDAx.py:111 in read_ndax
>     step_df = read_ndc(step_file)
> 
>   File C:\ProgramData\conda-venvs\py39\lib\site-packages\NewareNDA\NewareNDAx.py:223 in read_ndc
>     return f(mm)
> 
>   File C:\ProgramData\conda-venvs\py39\lib\site-packages\NewareNDA\NewareNDAx.py:394 in _read_ndc_11_filetype_7
>     rec.append([Cycle+1, Step_Index, state_dict[Status]])
> 
> KeyError: 22
> 
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> 
>   File C:\ProgramData\conda-venvs\py39\lib\shutil.py:627 in _rmtree_unsafe
>     os.unlink(fullname)
> 
> PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\ANDRSI~1\\AppData\\Local\\Temp\\tmptnhnrn6f\\data_step.ndc'

I am running very similar cycling protocols on other cells, which I get no error when processing. Upon comparison the only difference is an "OCV"-step, which lead me to believe that that is the culprit. What gives?

Grimler91, 2025-01-16:
Hi, 

There is indeed no step type 22 defined: https://github.com/Solid-Energy-Systems/NewareNDA/blob/master/NewareNDA/dicts.py#L44-L45 so that might very well be an "OCV step". Could you try adding it to dicts.py and see if you get the correct result? 

Out of curiosity, how does the OCV step work? What's the difference between OCV and normal Rest?

AndreSihm, 2025-01-16:
Yep, that was the issue. Added 22: 'OCV', to state_dict. 

Now I feel silly for not looking deeper haha. Thanks!

It's a rather useless step really, all it does is create an OCV flag within the protocol/data that can then later on be used as an identifier. It cannot be configured in any way (as far as I know). In other words: in a 'Rest' step you can set the time you want to rest, in an 'OCV' step you cannot.

Grimler91, 2025-01-16:
Aha, sounds like one of those kind of meta step types (like control, pause, SIM etc) 

Thanks for testing! I opened #101 to get the fix in.

d-cogswell, 2025-01-16:
Thanks @AndreSihm for explaining what the OCV step is, and @Grimler91 for submitting a fix.